UNH-OE/wave-tow-tank #39: Crack in beach gantry jack wheel(s)

petebachant, 2015-01-19:
At least one of these is cracked--see below. See #13 for some ideas for improving the system. 

[image]

petebachant, 2015-03-06:
This became a bit worse today. I recommend addressing #58. 

[image]

[image]

amcleodUNH, 2015-03-07:
One sick day and the whole world falls apart ….

Let’s consider changing over to hard rubber with bearings.

[image]<https://cloud.githubusercontent.com/assets/4604869/6519936/5a47d604-c389-11e4-8abe-322f88e1f29b.png>

[image]<https://cloud.githubusercontent.com/assets/4604869/6519912/177a3114-c389-11e4-83a1-b4d42c6be892.png>

petebachant, 2015-03-07:
It might be helpful, while this is being worked on, to add some type of track for the wheels so a single person could move the gantry. Some v-groove wheels and a track made from angle iron would be pretty cheap.

petebachant, 2015-03-16:
Fixed on March 13.

closed 2015-03-16 by petebachant