UNH-OE/wave-tow-tank #58: Modify beach gantry for single person operation

petebachant, 2015-03-02:
* Extend pendant cables ~8 ft so they can both be held simultaneously by one person.
* Remove swiveling trailer jacks add wheels/tracks (angle sections with V-groove wheels, e.g. http://www.mcmaster.com/#2310t692/=ujhhj1). Mount in threaded inserts in the floor so they can be removed--mainly important for the sections beyond the tank.

petebachant, 2015-03-06:
Picturing angle welded to flat plate screwed down into some McMaster 93980A530 or 93980A110, depending on if we need corrosion resistance to hardware embedded in the floor. Once again, a section that goes beyond the tank walls can be removed and only installed when it's necessary to pick up things like the turbine test frame.

Existing gantry 2 X 4 horizontal member could be cut out to fit wheel and axle. 

Tracks should have mechanical stops so wheels don't roll off the end.

petebachant, 2015-03-11:
Looking like the existing trailer jacks have 1000 lbf capacity based on http://www.harborfreight.com/automotive-motorcycle/trailer-jacks/1000-lb-swing-back-trailer-jack-41005.html

petebachant, 2015-03-11:
Correction: The actual jacks say 750 lbf each.

petebachant, 2015-03-11:
I propose we buy 4 of these wheels and weld them to the existing frame: http://www.mcmaster.com/#8741t63/=w9mf5d

I'd need to double check that we have an extension cord for the MIG welder to get it out into the high bay.

These whees can be used without tracks for now, but tracks can and probably should be added later.

petebachant, 2015-03-12:
A snapshot of what I'm talking about doing:

[image]

wosnik, 2015-03-12:
I’m on board with that, makes it easier to lift things.
Two questions need to be answered before you proceed:

-          Is the hoist ever used anywhere else in the lab (off-track)? Then wheels that turn would be required. Example: picking up the full beach that was assembled elsewhere in the lab.

-          How far would the angle iron tracks protrude over the end of the tank, and is that an issue with other stuff moving through the lab?
Check with Ken.

[image]<https://cloud.githubusercontent.com/assets/4604869/6610622/47d5e3b8-c838-11e4-99d1-19b6ec231bf7.png>

petebachant, 2015-03-12:
I don't think the hoist has ever been used elsewhere. There is another one around that could be used for that. Swivel casters have the same footprint, but are a little more expensive. I am more worried that they would prevent one person operation. 

I'm not planning on building the tracks right now, but when they were built, I imagine they would be removable. Users would install extensions to move beyond the beach, but normally that range of motion would be blocked off.

petebachant, 2015-03-16:
Implemented this on Friday. Going to close and open new issues for cable hooks and tracks.

closed 2015-03-16 by petebachant