UNH-OE/wave-tow-tank #59: Add tracks for beach gantry casters

petebachant, 2015-03-16:
Requirements:

* Easy to install and remove
* Inexpensive (angle iron)
* Have stops at the ends
* Allow for extension beyond the tank walls for picking up e.g. turbine test bed frame

petebachant, 2015-03-16:
Preliminary sketch of an idea below. Want to add bars for fitting sections into each other, mounting slots, and possibly make wide enough that they touch tank walls to make alignment easy.

[image]

petebachant, 2015-03-18:
# Fleshing out the design a bit

Angle is welded to flat plate with holes in it. At the end smaller pieces of angle are added to pull track sections together and eliminate gaps. 

The idea is that track sections would be screwed down using threaded inserts epoxied into the concrete floor. We don't want to use studs so the floor will remain flush when track sections beyond the tank are removed. 

[image]

## Connecting track sections

3 bars welded to the inside of the angle fit into the next section.

[image]

[image]

## Stops

These have 1/4-20 tapped holes so they can be easily removed to install additional track sections.

[image]

petebachant, 2015-03-18:
We probably want chocks for the wheels too, so e.g. when lowering the beach, the gantry will not be free to roll.

amcleodUNH, 2015-03-19:
How far beyond the end of the tank will this assembly reach?

petebachant, 2015-03-19:
In the "normal" configuration, the tracks would not extend beyond the tank. However, when I wanted to pick up turbines, for example, an additional set of of track sections could be installed. These would also keep the wheels from rolling into the floor drains.

petebachant, 2015-05-13:
CAD files for what I have so far (probably won't work on this much more, at least for a while):

* [SolidWorks 2015](https://drive.google.com/file/d/0BwMVIAlxIxfZX1lUa1hCcXhFUUE/view?usp=sharing)
* [STEP](https://drive.google.com/file/d/0BwMVIAlxIxfZbEV4bVpYRnpybFk/view?usp=sharing)