UNH-OE/wave-tow-tank #63: Upgrade PC to Windows 10

petebachant, 2015-07-30:
May want to make sure the ACS software is compatible, but it probably is.

Toby-D, 2016-07-27:
The free upgrade to Windows 10 ends on July 29th, I think. Should we go ahead and upgrade the tow tank computer?

amcleodUNH, 2016-07-27:
Have we tested all operational software compatible with Windows 10?

jcahern, 2016-07-27:
I have not tested the software.  extended support for win7 has another 5 years.  this will only include fixes for security risks.  I would expect the university would have a site license to win10 at some point in the next 5 years.  I'd like to see it upgraded but not sure two days is enough time to review the impact.

wosnik, 2016-07-27:
I am reluctant to upgrade, since this will likely cause compatibility issues with some of the drivers.
Pete: what is your opinion on this?

petebachant, 2016-07-27:
I'd stick with Windows 7 for now just to avoid breaking anything.

closed 2017-12-11 by petebachant