UNH-OE/wave-tow-tank #67: Address hydraulic ram interference

Toby-D, 2015-10-01:
After the fiberglass paddle was installed, the back end of the hydraulic ram makes contact with the green steel structure during each wave cycle. 
Possible causes: The paddle attachment point could be slightly higher than the original.

petebachant, 2015-11-06:
I think this was solved by Kevan Carpenter. The string pot was corroded, so it was replaced and sealed.

petebachant, 2015-11-10:
Note how the housing at the end of the cylinder is hitting the tray:

[image]

Essentially what needs to happen is that the end of the ram with the spherical joint needs to be lowered:

[image]

amcleodUNH, 2015-11-10:
Or the tray assembly raised.

[image]<https://cloud.githubusercontent.com/assets/4604869/11069380/de63610e-87a3-11e5-8f39-23338c9f8750.png>

[image]<https://cloud.githubusercontent.com/assets/4604869/11069438/1b21f358-87a4-11e5-9eda-e0cc6f4e8208.png>

petebachant, 2015-11-10:
My idea would be to replace the pivot bracket with a spherical bearing:

[image]

Examples from McMaster: http://www.mcmaster.com/#2433k16/=zr20tr

Not sure if they'd have a size that would drop right in.

petebachant, 2015-11-10:
Also maybe put slots in the pivot bracket mounting bracket (on the paddle) so there is some vertical adjustment. Maybe mill it down at bit as well.

amcleodUNH, 2015-11-10:
I like this idea because it would constrain some, not all, of the paddle twist.

[image]<https://cloud.githubusercontent.com/assets/4604869/11069677/5bbe401e-87a5-11e5-8302-9186532acd73.png>

petebachant, 2015-11-11:
I whipped up a simple CAD model to show that the existing spherical joint actually does provide too many degrees of freedom:

[image]

If you fix the paddle and the cylinder mounting bracket, the piston can still rotate about the spherical joint.

For my proposed change, the clevis bracket will need to be a bit wider than the spherical bearing, such that the pin can slide in the cross-tank direction to account for any misalignment there. 

The CAD assembly can be copied/viewed at Onshape: https://cad.onshape.com/documents/7ccbc3fe09744ef2808c3692/w/080b579ac67b403f9be877c8/e/f733d208fe4a45e2be403ca3

jcahern, 2016-04-06:
how does adding a spherical bearing prevent rotation in the yaw direction.  isn't it intended to allow yaw the way you are installing it?  and isn't yaw what we are trying to constrain to keep the paddle in a consistent across tank orientation so that waves are uniform.  the two issues that we are dealing with are preventing the ram from bottoming out on its mount plane, preventing the piston itself from flexing (which is why the spherical joint was added), and keep the paddle straight.  it seems the first two can be solved by changing the relative heights of the two attachment points of the ram and removing the spherical joint.  also replacing the ram with one of larger diameter will help with flexing. if we do those two things we eliminate the freedom of rotation for the paddle in the yaw direction.  however I don't think this is the dominant factor in why the paddle does not stay in the same across tank plane.  it appears to be mostly do to flexing of the paddle itself along the diagonal.  you can actually displace the corners by hand with 20-40lbs of force with like a 4" amplitude.  and this is not due to the ram attachment rotating.  so while the attachment can and should be improved I think the real issue is flex in the paddle.  do you all have suggestions to what reinforcement would look like?  it mainly needs to be constrained along the diagonals to each corner so an X pattern is most important.  so maybe a truss system that's beefy on the diagonals but prob also includes vertical member along each edge.

jcahern, 2016-04-06:
and by the by, I removed the wipers to see how much unbalanced friction with the sides of the tank was affecting yaw and it didn't seem like much.  the real issue is the feedback forces on the paddle as its moving.  because the system is not stiff enough, mostly in the paddle, also in the attachment, once these forces become unbalanced in an across tank direction, the paddle starts to yaw.  and this yaw creates more unbalanced forcing that influences the yaw.  once it sets up you pretty much can't get rid of it.  and unbalanced forces are inevitable because you are pushing at one point and are free on either end.  a dual ram system could also help with all of these issues including the flex.

petebachant, 2016-04-06:
I agree. The spherical joint replacement idea was to address the interference/underconstraint problem, not the yawing. The yaw/flex would need to be fixed by stiffening the paddle structure, which as you said would probably be best done with some trusses added to the back. I don't have any idea on the best configuration though. Maybe if we could get an FEA model whipped up that could predict the resonant frequency in that mode, we could use it to evaluate some ideas. 

We may be able to build on this example: https://www.simscale.com/projects/ahmedhussain18/truss_bridge_eigenfrequency_analysis

jcahern, 2016-04-06:
sweet.  its been 15yrs since i took tsukrov's class so I'll be way too slow.  I may see if he has someone that can help.  I know you guys are too busy.  I hope to brush up on this stuff again eventually.  maybe the timelines will coincide.  but I think the main problem is the flexing and is my current focus. besides the immediate problem of getting the beach back in.

jcahern, 2016-04-06:
can i use aluminum?

petebachant, 2016-04-06:
The chlorine in the tank generally corrodes aluminum pretty quickly, but Matt Rowell seemed to have some luck with some sort of powdercoating--I'm not sure of the details (relevant to #72). I think stainless and fiberglass are less risky.

amcleodUNH, 2016-04-06:
Even powder coated or anodized aluminum will be problematic. Use stainless.

jcahern, 2016-04-06:
?word

jcahern, 2016-04-07:
some really nice waves in this vid.  worth doing some research to see how its done

https://www.facebook.com/ScienceNaturePage/videos/808348552630738/