UNH-OE/wave-tow-tank #70: Replace upper belt sliding surface

petebachant, 2015-10-10:
From #65. The UHMW tape is worn out in some areas, which is causing the belt to make a lot of noise during towing. The wear is probably related to the fact that in some spots the tensioning clamp bracket rubs a little (#69 would probably address this). I am going to get a quote on a thicker continuous UHMW strip for the upper sliding surface. This should last quite a while longer, though we may need to install it with sheet metal screws if we can't get adhesive-backed strips that thick.

petebachant, 2016-01-30:
cc @wosnik @iangagnon 

Here's a closeup of the timing belt housing cross-section. The part that needs to be replaced is a 0.020" thick piece of adhesive-backed UHMW plastic film:

[image]

The tape is nominally 1245.5" long 4.75" wide, and the gap is designed to be 0.133" tall. We could probably step up to 0.060" thick strips or so. The width of the housing is 5.160", so we could go up to 5" or so wide on the new UHMW, though it's not necessary that it be wide.  

The original pieces (there is a lower strip as well) cost $537.

petebachant, 2016-01-30:
Could get a quote from these guys: http://www.eplastics.com/slick_strips

petebachant, 2016-02-01:
Steps to perform the repair:

1. Remove all outer timing belt housing pieces. In the picture above, these are the aluminum angle sections (one is polycarbonate, as the wavemaker end) in the upper right. Note that it will be easiest to keep these in order, despite being theoretically identical.
2. Release belt tension. See https://marine.unh.edu/oelab/wiki/doku.php?id=tow_tank:maintenance:procedures#setting_drive_belt_preload
3. Remove timing belt tensioning clamp ends and move the belt out of the way of the upper sliding surface.
4. Remove timing belt tensioning bracket from carriage via the four 1/4-20 hex bolts.
4. Peel off old UHMW film.
5. Clean surface with mineral spirits or acetone.
6. Apply new UHMW film. This usually comes with pressure sensitive adhesive so adequate pressure will need to be applied. I used a small plastic roller the first time. 
7. Re-install belt tensioning bracket at the carriage. Care should be taken to ensure the bracket is parallel with the carriage path.
8. Re-install belt tensioning clamps in the bracket and set preload according to https://marine.unh.edu/oelab/wiki/doku.php?id=tow_tank:maintenance:procedures#setting_drive_belt_preload
9. Re-install timing belt housing pieces. 
10. Test to ensure the belt resonance is gone.

petebachant, 2016-02-09:
Another Idea I had a while back was to put small ball bearings on the tensioning clamps or their mounting bracket to keep them from rubbing on the sliding surface as well, which is likely a cause of some, if not all of the wear. 

Maybe something like this: http://www.mcmaster.com/#2342k183/=111thav

[image]

petebachant, 2016-02-09:
I'm going to put together a drawing for this with part numbers on it. We will still need a quote for the UHMW though.

petebachant, 2016-02-09:
I can't seem to find the name of the manufacturer I bought the first rolls from--just the price. I think the [ePlastics](http://www.eplastics.com/slick_strips) guys will make custom rolls though.

petebachant, 2016-02-09:
From @jcahern:

>these guys will do 108' long roll, 5" width, 0.022" thick including acrylic adhesive for $436.  sound good?  they do 30 and 60mil but not in a length long enough.  they have it in stock so I can have it here next week.

>http://catalog.cshyde.com/category/ategories-tapes-with-adhesive-uhmw-tapes-1

With a 5" wide tape, the new rolling ball bearings will be rolling on the tape--not sure if that's a good or a bad thing! It's probably okay.

petebachant, 2016-02-09:
@jcahern, Do you want to try to get a quote from eplastics? I have a feeling they can do 4.75" wide, 0.060" thick, and the proper length if we ask.

jcahern, 2016-02-09:
> Bug from

prob worth avoiding.

petebachant, 2016-02-10:
Here's a drawing I whipped up for my proposed modifications:

[Belt tensioner mods.PDF](https://github.com/UNH-OE/wave-tow-tank/files/124050/Belt.tensioner.mods.PDF)

jcahern, 2016-02-10:
do you think the bearings should be spring loaded so when they do 
contact the rail there's some shock absorption?  maybe just some kind of 
polymer wheel?

petebachant, 2016-02-10:
I can't find any wheels small enough to fit in such a tight spot, and the plastic bearings I've seen are pretty low capacity and unsealed. The 6 x 6  beam that the tensioner mounts to actually does act like a spring as it twists, which is why I had to add the tie rod that connects the carriage face to the large I-beam. I actually think going with the bearings but getting the 5" wide tape would be optimal, so the wheels ride on a plastic surface. Let me know what you think.

petebachant, 2016-02-10:
And here's a snapshot of the rev 0 drawing for the sake of posterity: 

[image]

jcahern, 2016-02-10:
cool, sounds good.  I'll keep looking for 0.060" thickness but if I 
can't I'll just go with 0.020.  it can just be a maintenance item. but 
if the tensioner scuffing problem is fixed 0.020 should last longer 
anyway.  thanks for doing all this work, makes my job easy.

petebachant, 2016-02-10:
Yeah, it will probably be fine. I think the tank got used a bit more often and at higher loads in my experiments than it will in the future. 

That reminds me: I just migrated over a rough draft of the maintenance schedules we were attempting to establish for the tank. Feel free to edit if you have any ideas: https://github.com/UNH-OE/wave-tow-tank/wiki/Maintenance-schedule

jcahern, 2016-02-10:
does github have a messaging service?

petebachant, 2016-02-10:
Only via specific issues in the [public issue tracker](https://github.com/UNH-OE/wave-tow-tank/issues). We could create a [Slack](http://slack.com) team for the lab for other messaging purposes.

petebachant, 2016-03-14:
@jcahern how did this go? I haven't been in the lab for a while. Is this issue ready to be closed?

petebachant, 2016-03-14:
Also a reminder to update the tow system status after this is finished: https://github.com/UNH-OE/wave-tow-tank/wiki

jcahern, 2016-03-14:
i think so.  covers are still off if you want to take a look.  i don't 
see any issues.  it was in use last week but at slow speed. the side of 
the flange towards the wave generator bends down and contacts the rail.  
but it is covered with uhmw tape so there's not much friction.

petebachant, 2016-03-14:
After tightening the button head screws for the tensioning clamps, did you give the large tensioning clamp nuts an extra quarter turn or so? That arches the ends of the plate up so they won't contact the sliding surface. 

When do you think the timing belt guards will be back on? I will try to stop in before and take a look.

jcahern, 2016-03-14:
I'll give that a shot.  I wanted to retension after use anyway.

guards can go back on whenever.  prob this week.

jcahern, 2016-03-15:
pete,

let me know when you inspect the tank and i'll put the covers back on.

petebachant, 2016-03-15:
Sounds good. I probably won't make it over there until Friday afternoon/evening though. Lots of thesis writing!

jcahern, 2016-03-15:
good luck.  I'll leave it off.

wosnik, 2016-03-15:
Qualitative observation: At low speeds (~0.2-0.3 m/s) the belt/carriage runs much quieter that it used to.

https://github.com/UNH-OE/wave-tow-tank/issues/70#issuecomment-196891962

petebachant, 2016-03-18:
I just got a chance to check out the repair and it looks like you guys did a great job. I did adjust one thing though. After tightening the button head screws on the tensioning clamps, I tightened the large ~23 mm nuts about an extra half turn so the plate is somewhat concave, so the ends don't touch:

[image]

I'd say it's ready to go all back together, then we can close this issue and set the tow system status back to operational!

jcahern, 2016-03-21:
Perfect. Thanks Pete.

[image]<https://cloud.githubusercontent.com/assets/4604869/13893447/ce3e3262-ed34-11e5-80aa-d549429b92bf.png>

petebachant, 2016-03-27:
May want to do some more thorough testing, but since the surface is replaced, I'm going to close this issue and update the status.

petebachant, 2016-03-27:
Also added an item to the table at https://github.com/UNH-OE/wave-tow-tank/wiki/Maintenance,-repairs,-changes.

closed 2016-03-27 by petebachant