UNH-OE/wave-tow-tank #78: Maintenance procedures images

petebachant, 2016-02-09:
Just uploading these here:

[image]

[image]

[image]

closed 2016-02-09 by petebachant

jcahern, 2016-02-09:
> <https://github.com/UNH-OE/wave-tow-tank/issues/78#event-544875979>.Web Bug
> from

did you buy a custom roll in the past?  seems like they like to go 12, 
6, 3, 2, 1, etc.  but not 4.75 or 5" width.  have you tried to buy a 
larger roll and cut it down?

petebachant, 2016-02-09:
Hey @jcahern, just an FYI: I have been adding replies in https://github.com/UNH-OE/wave-tow-tank/issues/70