UNH-OE/wave-tow-tank #83: End timing belt cover missing screw

petebachant, 2016-04-25:
Looks like it was tough to get in (sometimes you can't do them sequentially so you have enough leverage to bend the ends). Should be able to get it with a little clamp though:

[image]

jcahern, 2016-04-25:
?thanks for the pic.  I actually delegated putting the covers on so I wasn't aware.  I'll give it a shot.

[image]<https://urldefense.proofpoint.com/v2/url?u=https-3A__cloud.githubusercontent.com_assets_4604869_14795510_d11c7eb4-2D0af6-2D11e6-2D9ece-2Db9fd5d0e5e66.png&d=CwMCaQ&c=c6MrceVCY5m5A_KAUkrdoA&r=abugFrCO37tu3CqHcsae0TtvpjoeINCQgnr3WWu12HE&m=U9-LxX1D0af6LmC8ZYX7kfw2QKir-c59DRZ_F7FR1dk&s=i7tOJXjjeIqP9Dwj2t0y3beQpBzcRYxxSmUVU9DMS0Y&e=>

-

petebachant, 2016-04-25:
Ah, I see. I would have thrown it in myself but didn't know where the screws are being stored.