UNH-OE/wave-tow-tank #95: Belt sliding surface needs to be repaired

petebachant, 2018-01-12:
May need to simply replace the tape, or design something a little thicker and install with sheetmetal screws.

petebachant, 2018-01-13:
May be a good idea to switch over to some sort of roller system. Thinking about ways to make this an easy retrofit.

petebachant, 2018-01-13:
Something like this could work: https://www.mcmaster.com/#2297t11/=1b3yqpe

jcahern, 2025-01-22:
the uhmw tape was replaced circa something like 2017 but quickly wore as the original did.  this does not appear to be a workable solution.  rollers might work or simply increasing the thickness of the low friction pad.  ie using thin plates of uhmw.  

removing the deteriorated tape is a significant undertaking.  mechanical scrapers and acetone was used.  I do not endorse using petroleum solvent if this work is repeated.  whatever the solution, thought should be paid to how the low friction elements are replaced in the future.  perhaps an embedded fastener system.  if adhesives are used they should be compatible with easy replacement.