UNH-OE/wave-tow-tank #100: Move CAD to Onshape?

petebachant, 2019-10-25:
@NicoleMarone this is mostly a heads up for you. If you haven't checked it out yet, Onshape is sort of like Google Docs for CAD, may now be mature enough to move away from using SolidWorks, and it's free for public projects. This would make collaboration a bit easier for those without SolidWorks licenses, and it's easy enough to export STEP files for others to do private development work in any CAD software.

jcahern, 2019-10-25:
Hey guys

I’m having some trouble with the network on the ttank computer. I know there were changes made when Pete was here. Please let me know what was done. 

Also in order to support the tank I need to know if changes are made. If I’m not included I will not be able to support the system. 

Thanks

petebachant, 2019-10-25:
To my knowledge nothing was permanently changed with the network. The WiFi adapter was giving trouble with the NI chassis so that was temporarily disabled, but we turned it back on before I left.