UNH-OE/wave-tow-tank #101: Ramp up tow carriage issue

NicoleMarone, 2020-02-07:
**Issue:** Occurs during testing with the test bed frame and turbine mounted on to the tow carriage at tow carriage speeds of 1.2 m/s or greater. After the turbine reaches the specified rpm, the carriage begins to accelerate to the specified speed (1.2 m/s or greater). During the acceleration of the tow carriage, the carriage jolts, making a loud noise and then comes to a complete halt. This occurs approximately 10 – 15 feet from the home position of the carriage, prior to the control platform. At tow carriage speeds less than 1.2 m/s, there is a noticeable, but very slight jolt of the carriage in the same location. However, the carriage continues to accelerate and reach the specified speed.  After the carriage was instantly halted while attempting to accelerate to 1.2 m/s, I reset the system and cleaned the linear encoder tape of debris. While there are slight imperfections on the tape in that location, these imperfections occur throughout as well. This has occurred with the UNH-RVAT on the test bed (1 m blades, 1 m diameter) and with another turbine (0.9 m blades, 0.45 m diameter). Cleaning the tape did not resolve the issue. When the test bed frame with the turbine is not on the carriage, the carriage does not have a problem reaching 1.2 m/s. It's almost as if the motor tries to reach the specified speed, overshoots, and then quickly shuts down.

**Fixes:** Possibly change ramp-up speed.

petebachant, 2020-02-08:
This sounds like an over-current fault in the motor, which would then cause the system to kill the motion. I forget exactly how, but there should be a way to read the fault history for that axis, probably through the ACS SPiiPlus app.

The jolt sound like it may be caused by the belt tensioner plate hitting part of the belt housing. You could take off a section of the belt housing cover and slowly move the carriage past that point to check for interference.

NicoleMarone, 2020-02-09:
Thanks for the help, Pete. I'll try to take a look into the history of the app to pinpoint the issue. We've looked at the belt tensioner in the housing while slowly moving the carriage over the area where the jolt occurs, and it doesn't appear that anything is interfering with it.  I'm thinking the jolt is from the system trying to slow down the carriage to the specified speed after accelerating past it, and that once it reaches past 1.2 m/s, the threshold to shut the system down is reached. Do you know if there a certain threshold?

petebachant, 2020-02-09:
That is a possibility--that it's an overshoot and the belt is losing tension to attempt to decelerate. It would probably be a good idea to reset the belt preload to spec: https://github.com/UNH-OE/wave-tow-tank/wiki/Maintenance-procedures#setting-drive-belt-preload

Beyond that, it is possible to tune the S700 servo drive to be slower/and or less aggressive at rejecting disturbances to speed. It's also possible to make the drive unstable if the tuning parameters are incorrectly adjusted, so it's not a procedure to be taken lightly! Try the belt tension first and see how that goes.

jcahern, 2025-01-22:
This was indeed an overcurrent issue that was verified in the alarm record.  the issue was traced to a bad disconnect switch on the outside of the junction box housing the kolmorgen carriage drive and motion controller.  the switch was leaking to ground and was replaced by an electrician.

closed 2025-01-22 by jcahern