UNH-OE/wave-tow-tank #103: makewaves app won't run

jcahern, 2025-01-22:
the makewaves desktop shortcut no longer points to a working incidence of makewaves.  the app appears to be running in the taskbar the gui does not appear.  this can be overcome by opening gitbash and navigating to /dev/makewaves under mingw64 and running the make command which creates a compiled instance of makewaves.

petebachant, 2025-01-22:
Sounds like maybe it's trying to restore its previous window location but it's off screen. You could try recreating the shortcut to the working version with: 

```sh
cd dev/makewaves
make shortcut
```

If that doesn't work, [this article](https://support.microsoft.com/en-us/topic/some-programs-are-displayed-off-the-screen-b2a915b7-4705-c455-febf-8ddb6e7f0ae4) shows how to move a window that's off screen, if that's the case.

jcahern, 2025-01-22:
yep, was offscreen.  I had to go to taskbar settings and enable taskview though.  now it stays where it supposed to be.  

I did the make shortcut command but couldn't find it.  is it supposed to land on the desktop?

tks Pete!

closed 2025-01-22 by jcahern

petebachant, 2025-01-23:
> I did the make shortcut command but couldn't find it. is it supposed to land on the desktop?

It should be created in the current working directory (`dev/makewaves`), so it can then be moved wherever.

jcahern, 2025-01-23:
I looked all over including there iirc.  would it show as the makewaves graphic or a generic icon?