UNH-OE/wave-tow-tank #105: risks of relative mode

jcahern, 2025-04-30:
potential exists to run carriage off of end of rails in relative mode.  this is a low risk problem for competent users and full control of the drive system poses some utility.  however, the carriage has been run off the end of the rails twice in the past few years apparently due to inexperienced operators running in relative mode when they should've been in global.  better controls to prevent inexperienced users from creating this problem should be implemented.  a formal documented training program conducted by vetted trainers would be an administrative control.  an engineering fix would be changing the software to put this feature behind an admin permission or simply eliminating it.

petebachant, 2025-04-30:
Eliminating the relative move feature would be quick. The system should probably always be homed and used in absolute positioning mode anyway. Ad hoc positioning can be done with jogging.

petebachant, 2025-05-08:
FYI I've started an issue in the Tow repo so I'll mention it here so they are linked: https://github.com/petebachant/Tow/issues/32?notification_referrer_id=NT_kwDOAEZDxbMxNjE5OTYxOTg3Nzo0NjA0ODY5#issuecomment-2863500523

petebachant, 2025-05-13:
@Pamboukes and I deployed a change today that forces Tow users to home the axis before doing any prescribed moves. This will allow the soft limits to engage if a user attempts to do a relative move that is out of bounds. Now that homing is required, it opens up the possibility of recomputing the relative move limits as a function of the current carriage position, so the user couldn't even select an out of bounds relative move if they wanted to. However, that has not yet been implemented.